kurbatov/firmata4j #7: Central Maven Repository is not updated

yferras, 2015-12-29:
Hello Oleg Kurbatov, 

Actually the latest version of your library in the Central Maven Repository is 2.3.2, but this version not supports SERVO mode. I reviewed the current code in github and I see that SERVO mode is supported. I think that if the code in github is stable, You should release a new version on Central Repository.

Thanks by your work, Ferrás.

kurbatov, 2015-12-29:
Hi Ferrás,

I'm going to release it between January 4 and 10. The release will include I2C support and this part should be tested before that. If you could test I2C from [this branch](https://github.com/kurbatov/firmata4j/tree/i2c), it makes the release sooner.

Thank you,
Oleg

yferras, 2015-12-29:
Thanks, I will test the i2c branch until You release the next version. 
Ferrás.

closed 2015-12-29 by yferras