kurbatov/firmata4j #12: Arduino resets the previous state

aliassad, 2016-08-15:
when i send digital message to arduino setdititalpin 8 high its work but when i send setdititalpin 9 high
the pin 8 goes low.
The arduino resets its self before sending a new message.

kurbatov, 2016-08-15:
Which version of firmata (.ino scetch) and firmata4j do you use?
Please, provide a code snippet to reproduce the issue.

aliassad, 2016-08-15:
Latest standardfirmata.ino and firmata4j 2.3.4

aliassad, 2016-08-15:
device.getPin(8).sevalue(1);
device.getPin(9).sevalue(1);

//the pin 9 gets high only

kurbatov, 2016-08-15:
Are those pins in output mode?

aliassad, 2016-08-15:
yes

kurbatov, 2016-08-15:
BTW, your snippet is not compilable.
Could you please, give some comprehensive snippet right from your code, where it is visible how do you know that something went wrong. Otherwise it's hardly possible to resolve the issue.

aliassad, 2016-08-15:
try{
             MainGUI.PORT_NAME=requestPort();
             
             device = new FirmataDevice(MainGUI.PORT_NAME); // construct the Firmata device instance using the name of a port
             device.start(); // initiate communication to the device
             device.ensureInitializationIsDone(); // wait for initialization is done        

             Pin p1=device.getPin(8);
             Pin p2=device.getPin(13);
             
                try {
         for(int i = 0; i < 1000; i++) {
            p1.setValue(1);
            Thread.sleep(500);
            p2.setValue(1);
            Thread.sleep(500);
             p1.setValue(0);
             Thread.sleep(500);
            p2.setValue(0);
            Thread.sleep(500);
         }
      } catch (IllegalStateException | IOException | InterruptedException e) {
         // TODO Auto-generated catch block
         e.printStackTrace();
      }

aliassad, 2016-08-15:
arduino  resets after each intstruction.

kurbatov, 2016-08-15:
Try to set `output` mode to the pins explicitly.

```java
    ...
    Pin p1=device.getPin(8);
    Pin p2=device.getPin(13);
    p1.setMode(Pin.Mode.OUTPUT);
    p2.setMode(Pin.Mode.OUTPUT);
    ...
```

aliassad, 2016-08-15:
Same Issue

aliassad, 2016-08-17:
I had attached LEDs on Arduino pins. the pin8 led gets low when the pin13 is high.

soundanalogous, 2016-08-17:
It appears the value of the port is not being tracked across writes. I've produced a crude (I don't write much Java) example to illustrate this:

```Java
package com.testportwrite;

import org.firmata4j.IODevice;
import org.firmata4j.firmata.FirmataDevice;
import org.firmata4j.Pin;

import java.io.IOException;

public class Main {

    public static void main(String[] args) {
        final IODevice device = new FirmataDevice("/dev/cu.usbmodem14231");
        try {
            device.start();
        } catch (IOException ex) {
            throw new RuntimeException(ex);
        }
        try {
            device.ensureInitializationIsDone();
        } catch (InterruptedException e) {
            e.printStackTrace();
            System.exit(1);
        }

        Pin pinA = device.getPin(8);
        Pin pinB = device.getPin(9);

        try {
            pinA.setValue(1);
            pinB.setValue(1);
            try {
                // keep LEDs on for 4 seconds
                Thread.sleep(4000);
            } catch (InterruptedException ex) {
                throw new RuntimeException(ex);
            }
            System.out.println("should have set pins");
        } catch (IOException ex) {
            throw new RuntimeException(ex);
        }

        try {
            device.stop();
        } catch (IOException ex) {
            throw new RuntimeException(ex);
        }

    }
}
```

The call to pinB.setValue(1) will set pinA back to 0. However, the port value should be maintained.

closed 2016-08-17 by kurbatov

kurbatov, 2016-08-17:
I have released fixed version 2.3.4.1

commit 3544724e (2016-08-17): Merge pull request #13 from kurbatov/fix/12

Fixes #12
commit aa2c106b (2016-08-17): Fixes #12
Error when calculating digital port value.