kurbatov/firmata4j #15: array index bound exception

myrdraal93, 2016-12-10:
There's a numerical overflow in "ask" method of FirmataI2CDevice class. The type byte in Java is signed. So, when the AtomicInteger return a value greater than 127, the value of variable "reg" is negative, and that, provoke an array index bound exception

kurbatov, 2016-12-10:
I have commited the fix for that. Could you please confirm if it works now?

myrdraal93, 2016-12-10:
now don't provoke an exception, but i can't get the response from the device when the index of array is greater than 127

myrdraal93, 2016-12-10:
a workaround may be to set the size of the array to 128. The instruction register.compareAndSet(256, 1) will become register.compareAndSet(128, 1)

kurbatov, 2016-12-12:
Unfortunately I haven't managed to reproduce the issue.

Another way around this bug is subscribing a listener and receiving updates continously. This way you don't have to invoke `ask` method all the time.

```java
    i2cDevice.subscribe(new I2CListener() {...});
    i2cDevice.startReceivingUpdates(messageLength);
```

kurbatov, 2017-06-08:
Register type was promoted from byte to int in 2.3.5 and it fixes described bug.

commit 7c082a43 (2016-12-10): Fixes #15
closed 2017-06-08 by kurbatov