kurbatov/firmata4j #16: Onewire function is needed

firefly917, 2017-02-15:
Thanks for your contribution of so great project.
Most functions of the project is enough to control the Arduino device, and it work well.
Now, I need use the temperature sensor of DS18B20 which is working on the protocol of Onewire.
How can I implements this function?  Would you add this function into this project?

kurbatov, 2017-02-15:
I have doubts on whether it is feasible to support Onewire in firmata4j
level at all.

Onwire's timings are too short for serial communication. Maybe newest
Firmata offers something to make it possible but I guess that Onewire
should be supported on firmware level i.e. if you tinker on standart
Firmata implementation before uploading it to your device.

However if you can find such Onewire support in other Firmata client
libraries in other languages, we could probably translate it to Java and
have the same functionality in firmata4j.

firefly917, 2017-02-15:
Thanks for your attention.
There is an implementation of Onewire at "http://npm.taobao.org/package/firmata-electron".
It is implemented by Node.js.  But I didn't try the library.

kurbatov, 2017-02-15:
I'll take a look at this when I have some spare time for firmata4j.

If you want this feature rather sooner, you can try implementing it on your
own. Firmata supports Onewire via sysex messages which are supported by
firmata4j. You can see the format of Onewire messages as Firmata defines
those here: https://github.com/firmata/protocol/blob/master/onewire.md

mchandia, 2018-01-13:
Hello, first of all many thanks for your contribution. Really appreciated :)
I'm implementing a onewire library (at least trying) but when I first set the mode for a pin as ONEWIRE in the firmata client application it says "Pin 6 doesn't support mode ONWIRE". Why could I be doing  wrong?, does the standard Firmata (loaded in arduino) support onewire ?

kurbatov, 2018-01-15:
Hi Mario,

I have read the [documentation on OneWire](https://github.com/firmata/protocol/blob/master/onewire.md) and it sais that [ConfigurableFirmata](https://github.com/firmata/ConfigurableFirmata/blob/master/examples/ConfigurableFirmata/ConfigurableFirmata.ino) got OneWire enabled by default. So it may be worth trying it with ConfigurableFirmata.

As for StandardFirmata I'm not sure whether it supports OneWire by default because I cannot find it in [the file that describes capabilities of different boards](https://github.com/firmata/arduino/blob/master/Boards.h).

mchandia, 2018-01-15:
Hello Oleg,

Thank you very much, I loaded ConfigurableFirmata into Arduino and it supports OneWire, so it works now :).

Mario

firefly917, 2018-03-11:
@kurbatov
Thanks very much!