kurbatov/firmata4j #19: Update protocol to latest version

centic9, 2017-03-06:
When using firmata4j with latest Firmata, we get a warning

> WARNING: Current version of firmata protocol on device (2.5) differs from version supported by frimata4j (2.3). Though these are compatible you may experience some issues.

Would be nice to have the protocol updated to the latest version.

I tried to take a quick look but unfortunately I did not understand some of the changes mentioned in https://github.com/firmata/protocol/blob/master/revisions.md and https://github.com/firmata/protocol/blob/master/protocol.md

kurbatov, 2017-03-06:
Hi Dominik,

I know that the protocol got newer versions since the time I originally developed **firmata4j**. New versions add new features that **firmata4j** currently does not support. Supported features should be working though.

I'm planning addition of OneWire support when I have some spare time because I have an OneWire powered sensor to test my implementation. Unfortunately I cannot provide implementation for every added feature because I don't have hardware to test it.

So I hope someone who sees the need in those features and understands them better will contribute their implementations eventually.

centic9, 2017-03-06:
Ok, understood, thanks for the explanation. I am not able to work on any of the new features myself, so there is probably not much use for this issue for now.

closed 2017-03-06 by centic9