kurbatov/firmata4j #24: Does this framework support COM ports?

Lighfer, 2018-05-04:
I want to use this framework to communicate two computer via serial port, does it support?    
I try to run follow demo:
```
 IODevice device = new FirmataDevice("COM6"); 
        
        device.addEventListener(new IODeviceEventListener() {
            public void onStart(IOEvent event) {
                System.out.println("Device is ready");
            }

            public void onStop(IOEvent event) {
                System.out.println("Device has been stopped");
            }

            public void onPinChange(IOEvent event) {
                Pin pin = event.getPin();
                System.out.println(
                    String.format(
                        "Pin %d got a value of %d",
                        pin.getIndex(),
                        pin.getValue())
                );
            }

            public void onMessageReceive(IOEvent event, String message) {
                System.out.println(message);
            }
        });

        device.sendMessage("aldjaklsdjaklsdjaklsdjklasd");
        device.stop();
```
But i got this error:
```
Exception in thread "main" java.io.IOException: Cannot send message to device
	at org.firmata4j.transport.SerialTransport.write(SerialTransport.java:87)
	at org.firmata4j.firmata.FirmataDevice.sendMessage(FirmataDevice.java:226)
	at org.firmata4j.firmata.FirmataDevice.sendMessage(FirmataDevice.java:215)
	at Main.main(Main.java:37)
Caused by: jssc.SerialPortException: Port name - COM6; Method name - writeBytes(); Exception type - Port not opened.
	at jssc.SerialPort.checkPortOpened(SerialPort.java:878)
	at jssc.SerialPort.writeBytes(SerialPort.java:348)
	at org.firmata4j.transport.SerialTransport.write(SerialTransport.java:85)
	... 3 more
```

kurbatov, 2018-05-18:
Hi Huang,

**firmata4j** uses jSSC (Java Simple Serial Connector) for working with serial ports. It supports COM ports.

The stack trace says that the port is not open. In order to open the port, `device.start()` method should be invoked before first `device.sendMessage()` invokation.

I would improve your sample the following way:

```java
    IODevice device = new FirmataDevice("COM6"); 
    device.addEventListener(new IODeviceEventListener() {
        public void onStart(IOEvent event) {
            System.out.println("Device is ready");
        }

        public void onStop(IOEvent event) {
            System.out.println("Device has been stopped");
        }

        public void onPinChange(IOEvent event) {
            Pin pin = event.getPin();
            System.out.println(
                String.format(
                    "Pin %d got a value of %d",
                    pin.getIndex(),
                    pin.getValue())
            );
        }

        public void onMessageReceive(IOEvent event, String message) {
            System.out.println(message);
        }
    });
    device.start(); // initiate communication to the device
    device.ensureInitializationIsDone(); // wait for initialization is done
    device.sendMessage("aldjaklsdjaklsdjaklsdjklasd");
    Thread.sleep(1000); // allowing some time for the device to respond
    device.stop();
```

Please, let me know if this solves the issue.

Sorry for delayed responce,
Oleg

Lighfer, 2018-05-26:
It works! Than you @kurbatov !

closed 2018-05-26 by Lighfer