kurbatov/firmata4j #27: Listed code in the "Usage" section does not work

Tf19, 2018-12-16:
I'm sorry if I do something completely wrong. But the following code does not work for me:

`public static void main(String[] args) {
		
		IODevice device = new FirmataDevice("COM8");
		try {
			device.start();
			device.ensureInitializationIsDone();
			device.stop();
		} catch (IOException e) {
			e.printStackTrace();
		} catch (InterruptedException e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		}
	}`

I know it should do nothing but connect to the Arduino anyway. But already the timer expires:

> java.lang.InterruptedException: Connection timeout
	at org.firmata4j.firmata.FirmataDevice.ensureInitializationIsDone(FirmataDevice.java:153)
	at arduino4j.Testklasse.main(Testklasse.java:21)

kurbatov, 2018-12-17:
That looks like specified port has a device on but that device doesn't respond to firmata messages.

Please, make sure that firmata is [uploaded](https://github.com/kurbatov/firmata4j#uploading-firmata-to-arduino) to the arduino board.

If you run this under linux (most probably not your case judging by the port name), your user may require additional permissions to interact with USB and serial ports.

If this doesn't help, please, provide more context as to what did you do, what did you try and where it went the unexpected way.

Tf19, 2018-12-17:
I use an Arduino Mega2560 running StandardFirmataPlus.

kurbatov, 2018-12-18:
Unfortunately I don't have a chance to test Arduino Mega with StandardFirmataPlus against firmata4j.

I have seen [an attempt to fix an issue with Arduino Mega](https://github.com/drjrkuhn/firmata4j/commit/29a8032141863b6f41dda877884baa5ca9fb6020#diff-22aca584af0351b3f06a59af8d16af25R426) among the forks. As far as I can recollect I backported it but that could be lost later on due to refactorings. I would appreciate if you check if that fix works for you and make a pull request if it does.

Another thing I can suggest is trying whether it works with just StandardFirmata.

Tf19, 2018-12-18:
Now I've tried to run it with StandardFirmata (instead of StandardFirmataPlus):

> Dez 18, 2018 4:05:47 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice handleEvent
WARNUNG: event error handler not implemented. will ignore event
Dez 18, 2018 4:05:48 PM org.firmata4j.firmata.FirmataDevice onProtocolReceive
WARNUNG: Current version of firmata protocol on device (2.5) differs from version supported by frimata4j (2.3). Though these are compatible you may experience some issues.
java.lang.InterruptedException: Connection timeout
	at org.firmata4j.firmata.FirmataDevice.ensureInitializationIsDone(FirmataDevice.java:153)
	at arduino4j.Testklasse.main(Testklasse.java:14)

I think only the last issues are interesting. From this I conclude that the communication now basically works. I will look at the link you posted later.

Tf19, 2018-12-18:
Now I've also implemented what's behind the link you posted: Now it works fine. Only the following errors are still being output:

> [firmata-parser-thread] WARN org.firmata4j.firmata.FirmataDevice - Current version of firmata protocol on device (2.5) differs from version supported by frimata4j (2.3). Though these are compatible you may experience some issues.

> Encoders not supported by firmware

But that should not be a problem.

By the way: It works with both StandardFirmataPlus and StandardFirmata.

kurbatov, 2018-12-18:
Hi @Tf19,

Could you please verify if the fix I applied to the master branch works for you?

firefly917, 2019-04-11:
This problem also happened when I used the board of Mega 2560 (ATMEGA16U2). Two different boards were all tested and they gave the same problem whether using the firmware of "StandardFirmataPlus" or "StandardFirmata".  

When I test the example of "Visualization" (https://github.com/kurbatov/firmata4j#visualization). Normally, the class of JPinboard should list all pins (totally 70 pins including 16 Analog pins and 54 Digital pins ) on the board. But it only list 64 pins (from NO.0 to NO. 63).

Two exceptions were reported.
----------------------------------------------------------------------
**1. "Connection timeout". This type of exception was always reported.**
```
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
java.lang.InterruptedException: Connection timeout
	at org.firmata4j.firmata.FirmataDevice.ensureInitializationIsDone(FirmataDevice.java:184)
	at Main.main(Main.java:20)
```
---------------------------------------------------------------------
**2. "java.lang.ArrayIndexOutOfBoundsException" and "Connection timeout". This type of exception was reported sometimes.** 
```
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Exception in thread "firmata-parser-thread" java.lang.ArrayIndexOutOfBoundsException: 1
	at org.firmata4j.firmata.parser.PinStateParsingState.process(PinStateParsingState.java:53)
	at org.firmata4j.fsm.FiniteStateMachine.process(FiniteStateMachine.java:146)
	at org.firmata4j.fsm.FiniteStateMachine.process(FiniteStateMachine.java:171)
	at org.firmata4j.fsm.FiniteStateMachine.process(FiniteStateMachine.java:157)
	at org.firmata4j.firmata.parser.FirmataParser$JobRunner.run(FirmataParser.java:91)
	at java.lang.Thread.run(Thread.java:748)
java.lang.InterruptedException: Connection timeout
	at org.firmata4j.firmata.FirmataDevice.ensureInitializationIsDone(FirmataDevice.java:184)
	at Main.main(Main.java:20)
```
----------------------------------------------------------------------------

I hope that these information could help to fix the problem.

Tf19, 2019-07-10:
Hi,

finally I found some time to try it again: I use again the Arduino Mega2560 (with both StandardFirmata and StandardFirmataPlus). In _org.firmata4j.firmata.FirmataDevice.java_ I edited the following lines starting at line 371:

`sendMessage(FirmataMessageFactory.pinStateRequest(pinId));
                    /* If the pin count is too high (i.e. Arduino Mega), then 
                     * too many firmata requests in a row can overflow the 
                     * device's serial input buffer. 
                     * One solution is to yield a little time between
                     * requests to allow the device to respond. The response may
                     * then safely sit in the host's much larger serial input
                     * buffer until it is dealt with by onPinStateReceive
                      */
                     try {
                         Thread.sleep(10);
                     } catch (InterruptedException ex) {
                         // ignore
                     }
                    
                    if (pinId > 0 && pinId % 14 == 0) {`

That's the [fix you mentioned above](https://github.com/drjrkuhn/firmata4j/commit/29a8032141863b6f41dda877884baa5ca9fb6020#diff-22aca584af0351b3f06a59af8d16af25R426).

Running StandardFirmata(Plus= I just get the following error:

> [firmata-event-handler-1] WARN org.firmata4j.firmata.FirmataDevice - Current version of firmata protocol on device (2.5) differs from version supported by firmata4j (2.3). Though these are compatible you may experience some issues.

I can do a pull request if you want, but you can also add those three lines by yourself (that would be faster I think)

kurbatov, 2019-07-11:
Hi @Tf19,

I didn't apply that exact fix because it introduced delay into initialisation of all boards no matter how many pins they had.

Since Arduino UNO has no problems initializing 14 pins in a row, I assumed that delaying after every 14 pin state requests should be enough even for bigger boards (the bigger a board, the better controller). However I have failed to pick the right duration of the delay as I don't have boards other than UNO.

So if it doesn't bother you too much, could you please tweak the duration of sleep at [line 381](https://github.com/kurbatov/firmata4j/blob/master/src/main/java/org/firmata4j/firmata/FirmataDevice.java#L381) and see if it will be sufficient to delay once in a while rather than every time?

Regarding the warning message you got: the board has a firmware that implements the newer Firmata protocol than firmata4j does. Those versions are compatible though firmata4j lacks support for the new features that were added in 2.4 and 2.5 versions of Firmata. Support of newer features is subject of the further development. Although everything that is available in firmata4j API should work fine.

Tf19, 2019-07-11:
The value for line 381 I ascertained now is at least 81 ms. But that's the absolutely minimum. Therefore, you should probably set the value a bit higher.
In addition, sometimes I get this error:

> [firmata-parser-thread] WARN org.firmata4j.fsm.FiniteStateMachine - Event handler is not registered for error:null. The event is ignored.

Sometimes it appears five times, sometimes eight times and sometimes it does not appear at all

kurbatov, 2019-07-11:
Thank you for the research. I'm going to set that 100 ms.

As for the missing event handler, it will probably require some further investigation and first of all improvement of the logging of that particular case.

closed 2019-07-11 by kurbatov

kurbatov, 2019-07-11:
I've pushed the fix for the delay and improvement for logging of missing event handler. Now I expect it will log additional info along the following lines:

```
  Unknown control token has been received. Skipping. 0xXX
```

where `0xXX` would be the byte that Arduino sends to the client.

Hope it will clairify the way the error should be fixed for good.

commit 7f305aca (2018-12-18): fix(firmata-device): initialization of Arduino Mega 2560

Add delay after every 14 pin capability requests in order to prevent
overflow of the input buffer on devices with numerous pins.
Execute event handling in a dedicated thread so that delays in handling
do not suspend processing of incoming bytes.

Fix #27
commit 60a7160d (2019-07-11): fix(firmata-device): increase dalay when sending pin state requests

Arduino Mega gets input buffer overflown with initial pin state requests

Fixes #27