kurbatov/firmata4j #29: Missing documentation for pin reporting

jpink, 2019-01-10:
Hi,

how can I set sampling interval or disable analog / digital pin value reporting? I see them in `org.firmata4j.firmata.FirmataMessageFactory`, but I don't know how to call it.

And why sampling interval must be less than 100 ms? Why not 5 min or longer? When i open (serial) connection I think there is too much unnecessary traffic.

kurbatov, 2019-01-10:
Hi @jpink,

You can set sampling interval the followng way:

    device.sendMessage(FirmataMessageFactory.setSamplingInterval(12));

For the reasons behind the threshold values, please, refer to the [protocol description](https://github.com/firmata/protocol/blob/f2fb835616876c221c2f5f80701c41ab6590a80e/protocol.md#sysex-message-format). I believe you can customize the implementation of a firmware so that it supports desirable sampling interval.

jpink, 2019-01-11:
[That method](https://github.com/kurbatov/firmata4j/blob/master/src/main/java/org/firmata4j/firmata/FirmataDevice.java#L260) seems to be package protected in latest release 2.3.6. I have use latest cloned one.

kurbatov, 2019-02-01:
Hi @jpink,

`sendMessage` method is part of public API in firmata4j-2.3.7

closed 2019-02-01 by kurbatov