kurbatov/firmata4j #33: OnCapabilitiesRecieved Missing

MarinaMatore, 2019-07-13:
We need a callback when  mFirmata.sendMessage(REQUEST_CAPABILITY); finishes.
Or when the pin list gets filled.

Or did I miss it somewhere?

Kind Regards

kurbatov, 2019-07-15:
Hi Marina,

You can achieve this using low-level message handlers:

```java
    IODevice device = new FirmataDevice(port);
    device.addProtocolMessageHandler(FirmataEventType.PIN_CAPABILITIES_MESSAGE, new Consumer<Event>() {
        @Override
        public void accept(Event evt) {
            //do the stuff here
        }
    });
    device.start();
```

That will basically add another handler to already registered one which you amend in #34. Since additional delay is required only for Adafruit Feather 32u4 board, I beleive it would be better to handle delays in such an additional handler.

kurbatov, 2019-07-15:
On the second thought the original design with delays is poorly implemented anyway. I think that all the pin state requests should be queued and sent only when response for the previous one is received. This approach yellds robust and performant solution.

I beleive the primary handler should be redisigned because this exact place generates new issues with noticable frequency.

MarinaMatore, 2019-08-10:
Hey!

When will your[ latest commit](https://github.com/kurbatov/firmata4j/blob/master/src/main/java/org/firmata4j/firmata/FirmataDevice.java) be available via Android gradle? Currently I still get the old commit.

Currently using: **implementation 'com.github.kurbatov:firmata4j:2.3.7'** 

Thanks
Best wishes.

closed 2019-08-26 by kurbatov

kurbatov, 2019-08-26:
Hi @MarinaMatore 

I have released 2.3.8 which contains update of pin initialization. It will be available soon via Maven Central.

commit 8839902e (2019-08-26): fix(init): initialize state of pins one by one

If the pin count is too high (i.e. Arduino Mega), then too many firmata
requests in a row may overflow the device's serial input buffer.
The solution is to request states of the pins one by one, i.e. request
the folowing pin's state after the previous responce is received.

Fixes #33