kurbatov/firmata4j #37: `setDigitalPinValue` method creates a `DIGITAL_MESSAGE` message instead of a `SET_DIGITAL_PIN_VALUE` message

HawaiianSpork, 2019-10-19:
This method is called `setDigitalPinValue` but it actually builds a `DIGITAL_MESSAGE` message:

https://github.com/kurbatov/firmata4j/blob/master/src/main/java/org/firmata4j/firmata/FirmataMessageFactory.java#L226

If you are using `StandardFirmata.ino` this can result in the wrong pin being changed since `SET_DIGITAL_PIN_VALUE` changes the value of a single pin where `DIGITAL_MESSAGE` changes the value for all pins on a port.

`DIGITAL_MESSAGE`
https://github.com/firmata/arduino/blob/master/FirmataConstants.h#L38

`SET_DIGITAL_VALUE`
`https://github.com/firmata/arduino/blob/master/FirmataConstants.h#L44`

kurbatov, 2021-01-05:
Support of that message was added in v2.5 of the protocol. Currently **firmata4j** targets 2.3.  We should add support of 2.4 features of the protocol before switching to that message to set pin's value.

kurbatov, 2023-05-23:
Previous attempt to fix it #59 was reverted in #63