kurbatov/firmata4j #42: JRE crash with GraalVM 20.1.0

NorbertSandor, 2020-09-01:
It is easy to reproduce: every call of `org.firmata4j.IODevice.start()` crashes the JVM:

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=12504, tid=6892
#
# JRE version: OpenJDK Runtime Environment GraalVM CE 20.1.0 (11.0.7+10) (build 11.0.7+10-jvmci-20.1-b02)
# Java VM: OpenJDK 64-Bit Server VM GraalVM CE 20.1.0 (11.0.7+10-jvmci-20.1-b02, mixed mode, sharing, tiered, jvmci, jvmci compiler, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#

---------------  T H R E A D  ---------------

Current thread (0x000002369e22a000):  JavaThread "main" [_thread_in_native, id=6892, stack(0x000000e67c200000,0x000000e67c300000)]

Stack: [0x000000e67c200000,0x000000e67c300000],  sp=0x000000e67c2fbfb0,  free space=1007k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [jSSC-2.8_x86_64.dll+0xb5db]

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J+0
j  jssc.SerialPort.openPort()Z+65
j  org.firmata4j.transport.SerialTransport.start()V+14
j  org.firmata4j.firmata.FirmataDevice.start()V+24

...
```

kurbatov, 2020-09-01:
It seems that `jssc` crashes. If jSerialComm that you propose in #41 doesn't fail on Graal, this issue get solved by implementing a `Transport` interface using working library.

I didn't mean supporting Graal when building the library but I would highly appreciate if you contribute a solution to this issue.

NorbertSandor, 2020-09-01:
The implementation proposed in #43 works with GraalVM as well.

dougmeredith, 2020-12-23:
It sounds like this is under control, but a further data point, just in case: Same issue with AdoptOpenJDK 11.0.7 on Windows.

dougmeredith, 2020-12-23:
A quick update: JSerialCommTransport  solved the problem for me too. Thanks!

Based on issues I had with an older version of nrjavaserial, I suspect that the original problem occurs on Windows with any JVM newer than 8.

centic9, 2021-01-04:
Not sure why, but I now have the exact same crash on Windows with pure OpenJDK 8 & 11, I am fairly sure it worked before without any update to the JDK in the meantime, so I don't know what actually changed here, maybe some Windows update broke jssc? 

The new serial transport from #43 works fine for me as well.