kurbatov/firmata4j #45: A fatal error has been detected by the Java Runtime Environment

Teloyrks, 2020-11-28:
StandardFirmata is uploaded on my arduino every thing is fine. but when i run the code eclipse shows 
[image]
and the creates this .log file
[hs_err_pid7000.log](https://github.com/kurbatov/firmata4j/files/5610157/hs_err_pid7000.log)

kurbatov, 2020-11-28:
Please, add the following line after `device.start()` and see what happens:

```java
device.ensureInitializationIsDone();
```

Teloyrks, 2020-11-29:
Still same :(
[image]

Teloyrks, 2020-11-30:
please help meh !!!!!!!!
please !!!!!
need more info ?

kurbatov, 2020-12-01:
It looks like a problem with a native part of `jssc` library.

There is one thing you can try with unreleased version of `firmata4j`.

The first step would be cloning this repository and building it locally:

```
$ git clone ggit@github.com:kurbatov/firmata4j.git
$ cd firmata4j
$ mvn install
```

After that you want to modify the `pom.xml` of your project with regard to dependencies to `firmata4j`:

```xml
<dependency>
    <groupId>com.github.kurbatov</groupId>
    <artifactId>firmata4j</artifactId>
    <version>2.3.9-SNAPSHOT</version>
    <exclusions>
        <exclusion>
            <groupId>org.scream3r</groupId>
            <artifactId>jssc</artifactId>
        </exclusion>
     </exclusions>
</dependency>
<dependency>
    <groupId>com.fazecast</groupId>
    <artifactId>jSerialComm</artifactId>
    <version>2.6.2</version>
</dependency>
```

Note that the version of `firmata4j` changes, `jssc` is excluded and a newer `jSerialComm` library is added to the dependencies.

Now, in order to use `jSerialComm` library, modify your code along the following lines:

```java
IODevice device = new FirmataDevice(new JSerialCommTransport("/dev/ttyUSB0"));
device.start();
device.ensureInitializationIsDone();
```

Note that the constructor of `FirmataDevice` should get an instance of `JSerialCommTransport` to use `jSerialComm` library as a transport to the device.

Teloyrks, 2020-12-05:
thanks. it worked :) <3

closed 2020-12-05 by Teloyrks