kurbatov/firmata4j #47: Incorrect values reported when using PULLUP mode

dougmeredith, 2021-01-28:
When using PULLUP mode for pins, they don’t always report the right value. This occurs with a genuine Uno, a clone Uno, and a clone Mega. It occurs on Windows, CentOS and on the Raspberry Pi.

I am running 2.3.8, although with the JSerialComm transport pulled from Git. I’m using Java 11. The Arduinos are running the latest version of StandardFirmata.

I tried the same test using a Python Firmata library, and didn’t have the problem, so I believe this is an issue with Firmata4J, rather than with the firmware. This is easy to reproduce, as described below.

**Basics**

- All tests are performed with two pins in PULLUP mode, and the other pins left with their default settings.
- The pin-under-test (PUT) is connected directly to a ground pin. Its value should therefore always be zero.
- No other pins have wires connected to them
- The second pin is called OTHER. As it is not connected, it should always report one.
- The results described here apply to pins 2 - 13. While this was tested on both an Uno and a Mega, no extensive testing was done on pins above 13.

**Test procedure**

- Call start()
- Call ensureInitializationIsDone()
- Set the mode of PUT and OTHER to PULLUP. Note: The order in which this is done matters.
- Wait 100 ms
- Get the values of the pins

**Test 1**

- Pin mode to set first: OTHER
- Constraints: . This should work using any two pins from 2 to 13
- Result
-      PUT. 1 (**incorrect**)
-      OTHER. 1 (correct)

**Test 2**

- Pin mode to set first: PUT
- Constraints: If PUT <= 7 then OTHER <= 7. If PUT >=8 then OTHER >= 8
- Result
-     PUT. 0 (correct)
-     OTHER. 1 (correct)

**Test 3**

- Pin mode to set first: PUT
- Constraints: If PUT <= 7 then OTHER >= 8. If PUT >=8 then OTHER <= 7
- Result
-     PUT. 1 (**incorrect**)
-     OTHER. 1 (correct)

kurbatov, 2021-01-28:
Hi @dougmeredith,
Thank you for reporting that. I will take a look at the issue.

dougmeredith, 2021-01-29:
Thanks! Just let me know if you need any more information, or if there is anything that you'd like for me to test.

oliviercoma, 2021-12-18:
I think it might be the same root cause as #55