kurbatov/firmata4j #48: Document how to use JSerialComm Transport Layer

philippegirard, 2021-04-21:
I am using firmata4j on Windows 10 with OpenJDK 11 and I get the same errors as @centic9 https://github.com/kurbatov/firmata4j/issues/42#issuecomment-753841966

I want to use JSerialCommTransport coded by @NorbertSandor here https://github.com/NorbertSandor/firmata4j/commit/cd9cb59bec502e2f027f33f9fac404da5a7c9398

However I am unable to import JSerialCommTransport with version 2.3.8. 
[image]

Maybe a 2.3.9 version including the changes should be released? How can I use JSerialCommTransport in the meantime?

I think there should be a section in the readme that explain how to use each Transport Layer interface (and how to install optional dependencies when needed). I am not very good with maven.

I am happy to help and contribute if I can!

NorbertSandor, 2021-04-21:
Until released, you can:
- add the `com.fazecast:jSerialComm:2.6.2` dependency to your build
- copy the https://github.com/kurbatov/firmata4j/blob/master/src/main/java/org/firmata4j/transport/JSerialCommTransport.java file to your project
- use `FirmataDevice(JSerialCommTransport(serialPortName))` to create the `FirmataDevice` instance

Let me know if the above steps are not sufficient.

philippegirard, 2021-04-21:
I was able to make it work with the steps you specified. Thank you @NorbertSandor 🎉

centic9, 2021-04-21:
I use the same workaround, but an official release with this new transport would be nice!

philippegirard, 2021-04-28:
I made an unofficial release of firmata4j with the latest commit from the master branch. https://github.com/philippegirard/firmata4j-autopack

You can find today's release here: https://github.com/philippegirard/firmata4j-autopack/tree/main/releases/2021.04.28

If you don't need the `jar-with-dependencies`, you can pick and choose dependencies from here: https://github.com/philippegirard/firmata4j-autopack/tree/main/releases/2021.04.28/separated

Works very well on Windows 10 and gives access to the JSerialCommTransport transport layer
[image]

If you new more recent releases in the future, just open an issue and it will automatically compile the latest commit in an new release.

closed 2021-04-28 by philippegirard