kurbatov/firmata4j #55: onPinStateReceive should not update pin value

oliviercoma, 2021-12-17:
According to the Firmata documentation, the pin state response do not return the value of the pin, but:
"For input modes, typically the state is zero. However, for digital inputs, the state is the status of the pull-up resistor which is 1 if enabled, 0 if disabled."

Updating the pin value in onPinStateReceive leads to incorrect values being reported by Firmata4j

kurbatov, 2021-12-18:
Thanks for reporting that. I'll take a look at this during new year holidays.

oliviercoma, 2021-12-18:
I'm currently working on a fix. I'll create a PR.

oliviercoma, 2021-12-18:
Also, the if
```
if (pin.getMode() == null) {
    pin.initMode(Pin.Mode.resolve((Byte) event.getBodyItem(PIN_MODE)));
}
```
is also an a potential issue I think. The device should be the owner of the mode, and if it reports the mode to be something (e.g. INPUT), it should not be filtered. It should probably also triggers the onModeChange event.

This issue will probably only happen if some custom code is added to the physical device (i.e. not StandardFirmata) that changes the pin mode directly. Unlikely to happen, but it could lead to a difficult to investigate issue

PR #56 (2021-12-31): fix 55: Do not update pin value from pin state response message as th…
fix #55 : Do not update pin value from pin state response message as the state in the message is not the actual value of the pin. See https://github.com/firmata/protocol/blob/master/protocol.md
closed 2021-12-31 by kurbatov