kurbatov/firmata4j #62: setValue in v2.3.9 not setting digital output correctly.

onnimikki, 2023-05-23:
Hi,

First, I need to say that I've enjoyed Firmata4j immensely and have been using it with my students here at York University.  Thank you for your work.

I'm testing out the 2.3.9 snapshot and it seems that the digital output setValue() method isn't working correctly.  I've tried it in two scenarios:

1. The PinBoard example 
2. My own example (https://www.yorku.ca/professor/drsmith/2022/02/25/easy-java-arduino-with-firmata/#BlinkingLED)

I've got an LED attached to pin D4 on my Arduino board.  Using the 2.3.8 release, it blinks fine.  However, in the 2.3.9 snapshot the board connects but the LED doesn't blink.  

I can confirm that the analog input seems to work on the A0 (pin 14) in the PinBoard example in the 2.3.9 snapshot, however.

James

kurbatov, 2023-05-23:
Hi James,

Thank you for reporting this issue. Probably this logic was broken in #59. I have merged it together with some other changes without proper testing in hope that it fixes #37.

I will roll back [7d89ad9](https://github.com/kurbatov/firmata4j/pull/59/commits/7d89ad9d5eb75e8e9fbf1c3f3acde98a621042ed) and look for another way to fix the issue.

I have seen your awesome blog posts on Arduino projects at York University some time ago and was pleased you used firmata4j. Thank you, sir, for your teaching and for publishing educational blog posts like this for people around the world.

With best regards,
Oleg

onnimikki, 2023-05-25:
Hi.  I ran an initial test with the updated files and, while the board connects, it doesn't seem to change the output state of one of the digital pins.  I have to put it aside for now and will come back to it and check again.
-J

PR #63 (2023-05-23): Revert "fix(pin): send correct message when changing digital output pin's value"
This reverts commit 7d89ad9d5eb75e8e9fbf1c3f3acde98a621042ed.

Fixes #62.
Reopens #37.